Severson-Group/AMDS #68: Add mechanical components to schematic and BOM

elsevers, 2025-05-13:
# Abstract
Add mechanical components to schematic and BOM for [Rev D release](https://github.com/Severson-Group/AMDS/tree/v2.0.x/Mainboard/REV20200920D)

# Context
We are in the process of updating our release folders for our various designs. Related issues for AMDS:
- #63 
- #64 
- #65 

@mohamed-dek1 has been working on adding mechanical components to schematics in Altium so that they auto populate on the BOM. For example:
- https://github.com/Severson-Group/AMDC-Hardware/pull/261
- https://github.com/Severson-Group/AMDC-Hardware/pull/259

We discovered that the current AMDS release files on the `develop` branch do not show the mechanical components on the schematic and BOM. 

# Approach

Close this issue via a PR that adds the mechanical components to the [rev D](https://github.com/Severson-Group/AMDS/tree/v2.0.x/Mainboard/REV20200920D) schematic and BOM.

mohamed-dek1, 2025-05-16:
per 05/16:

The mechanical components used to hold up the board also need to be added to the schematic and BOM.

elsevers, 2025-05-17:
## Update on Standoffs

1. The AMDS needs two different stand-off solutions. One for the main board to be lifted off of the table and one for the sensorcards.
2. The AMDS mainboard's currently released [BOM](https://github.com/Severson-Group/AMDS/blob/v2.0.x/Mainboard/REV20200920D/SensorMotherBoard_BOM_10qty.csv) lists only one stand-off:
    - [732-10550-ND](https://www.digikey.com/en/products/detail/w%C3%BCrth-elektronik/970110321/6174770?s=N4IgTCBcDaIOwGYwFoCMAGArJ9yByAIiALoC%2BQA): 11mm M3 stand off (and associated M3 screws `36-7689-1-ND`). I think this is for holding the mainboard off the table, NOT for the sensor cards 
    - The AMDC BOM uses 4-40 , 1/2" standoffs. **Let's switch the standoffs (and screws) to hold the mainboard off of the table to match [the  AMDC](https://github.com/Severson-Group/AMDC-Hardware/blob/develop/REV20231005F/bom-rev-f.csv)** so we have fewer parts to stock.
        - After making the switch, we'll need to update the [Google ME 76 Hardware Sheet](https://docs.google.com/spreadsheets/d/16eWoWGY_jfMSNuZQeauz20iACJenNDp30TCMVM3sQI8/edit?gid=1879693648#gid=1879693648)
2. So what are the correct stand-offs for the sensor cards?
    - The [current](https://github.com/Severson-Group/AMDS/blob/v2.0.x/CurrentCard/REV20200830C/CurrentSensorBoard_BOM.csv) and [high voltage](https://github.com/Severson-Group/AMDS/blob/v2.0.x/HighVoltageCard/REV20200913C/VoltageCard_bom_qty_10.xlsx) cards do not indicate any standoffs in their BOMs. 
    - The [low voltage card](https://github.com/Severson-Group/AMDS/blob/v2.0.x/LowVoltageCard/REV20200902B/BOM_qty1.csv) is quite confusing. The [BOM](https://github.com/Severson-Group/AMDS/blob/v2.0.x/LowVoltageCard/REV20200902B/BOM_qty1.csv)...
        - description refers to a 4-40 standoff / screw (great! this matches the AMDC!)
        - but... the standoff part number provided ([732-10538-ND](https://www.digikey.com/en/products/detail/w%C3%BCrth-elektronik/970090321/6174758?s=N4IgTCBcDaIOwGYwFoCMAGArAgHMgcgCIgC6AvkA)) points to an M3 9mm standoff. Oy..........
    - @noguchi-takahiro and I tried to use one of the AMDC 4-40 1/2" stand-offs for these boards but discovered that this is too long, the sensor card will not seat properly on the AMDS
    - Recently, we have converged on using a 4-40 3/8" stand-off for the Kaman I/O board. See this:
        - https://github.com/Severson-Group/Lab/issues/309#issuecomment-2831355635
        - It would be wonderful if we could re-use these stand-offs for the sensor cards too. This would reduce the number of unique parts we must stock.

## Next steps

@mohamed-dek1, please:
1. Figure out the correct stand-offs for the sensor cards. Try using the [4-40 3/8" stand-off parts we are now using for the Kaman I/O board](https://github.com/Severson-Group/Lab/issues/309#issuecomment-2831355635) to connect a sensor card to the AMDS. Does this seem like a reasonable solution? How does the height compare to the current stand-offs that are used? Take a picture / show me it.
2. Figure out our strategy for what stand-offs go on what BOM. For sure the main-board standoffs (to lift it off the table) go on the main board's BOM and schematic. But where should the sensor card standoffs go? They could reasonably go on either the main board's material or the sensor cards. However, putting them on the sensor card's is duplicative, since we have three sensor card designs. **Therefore, please indicate all stand-offs** (those to lift the main board off the table and those in between the sensor cards and the main board) **on the main board schematic + BOM.**
3. Create your PR to close this issue.

mohamed-dek1, 2025-05-19:
## 4-40 3/8" Stand-Off (proposed)

[image]

[image]

## M3 11mm Stand-Off with M3 6mm Screw (current)

[image]

[image]

##
There's a gap between the board and standoff when using the 4-40 3/8" stand-offs, however, when using the M3 11mm stand-offs, the jumper with fewer pins doesn't reach the ground fully. I spoke with @noguchi-takahiro and think that we should continue using our existing stand-off solution for mounting the sensor boards.

##
I agree that all standoffs and screws should be on the mainboard schematic + BOM. One thing I noticed was that there is another component that holds the sensor cards to the mainboard.

[image]

I think we should use 2 screws on either end of the stand-off such that they are short enough as to not hit each other and still fasten the sensor card and mainboard to each other.

[image]

mohamed-dek1, 2025-05-20:
per 05/20:

@knvprasad-dr and I have looked into the mounting setup for the sensor boards and think that using one 11mm M3 standoff and 2 6mm M3 screws are sufficient in mounting the sensor cards to the mainboard. Due to the thickness of both PCBs, we are able to get away with using 2 6mm rather than shorter ones. 

11mm standoff + 2 * (1.6mm thickness boards) = 14.2mm

14.2mm - 2 * (6mm screw) = 2.2mm of clearance between screws within the standoff.

elsevers, 2025-05-22:
Thanks @mohamed-dek1. Good work here and I think we'll need to chat to finalize details on this. 

One request in the meantime: can you identify an option for these standoffs in imperial sizing that use a 4-40 thread? I.e., maybe 7/16" length? Since we are using imperial stand-offs for the AMDC, I am thinking we should discuss what it would mean to standardize on that across all stand-offs so that we have fewer parts to stock

(CC @knvprasad-dr regarding the full-bridge converter)

mohamed-dek1, 2025-05-23:
per 05/23:

add [1772-3771-ND](https://www.digikey.com/en/products/detail/raf-electronic-hardware/2056-440-SS-20/9836151) to the [ME 76 Hardware Sheet](https://docs.google.com/spreadsheets/d/16eWoWGY_jfMSNuZQeauz20iACJenNDp30TCMVM3sQI8/edit?gid=1879693648#gid=1879693648)

mohamed-dek1, 2025-06-09:
@elsevers, what chart string should I use to order the standoffs?

mohamed-dek1, 2025-06-09:
Standoffs are being ordered in the following issue:

- https://github.com/Severson-Group/Lab/issues/341

PR #76 (2025-06-20): Update schematic and BOM to include screws and standoffs
This PR adds mechanical components to the AMDS REV D schematic and BOM. These components are used to:

-  Hold the mainboard up
- mount the sensor cards to their slots

##
this pr closes #68
closed 2025-06-23 by mohamed-dek1